Xinyuan-LilyGO/LilyGo-EPD-4-7-OWM-Weather-Display #5: T5 4.7 Inch E-paper V2.3 ESP32-S3 ?

copilador, 2023-08-30:
Hi.
It is valid to use the T5 4.7 Inch E-paper V2.3 ESP32-S3 https://www.lilygo.cc/products/t5-4-7-inch-e-paper-v2-3

Thank you

copilador, 2023-09-02:
OK Tested, works correctly

closed 2023-09-02 by copilador

NKCorpv2, 2023-11-01:
Dear Author. I've seen you file and it looks amazing. I've bought the T5 4.7 Inch E-paper V2.3 ESP32-S3, but I'm not able to load it. I have error messages from ArduinoJson. I am new to this, but I'd like to learn and try this code on my ESP32-s3. Can you help me? THank you in advance,

NKCorpv2, 2023-11-01:
error: cannot bind non-const lvalue reference of type 'std::__cxx11::string&' {aka 'std::__cxx11::basic_string<char>&'} to an rvalue of type 'std::__cxx11::string' {aka 'std::__cxx11::basic_string<char>'}
     convertFromJson(src, result);  // Error here? See https://arduinojson.org/v6/unsupported-as/

:\xxx\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\include\c++\8.4.0\bits\basic_string.h:514:7: note:   after user-defined conversion: 'std::__cxx11::basic_string<_CharT, _Traits, _Alloc>::basic_string(const _CharT*, const _Alloc&) [with _CharT = char; _Traits = std::char_traits<char>; _Alloc = std::allocator<char>]'
       basic_string(const _CharT* __s, const _Alloc& __a = _Alloc())

These are some errors I receive.
Thanks a lot!

NKCorpv2, 2023-11-01:
The library I am usign is the last update 6.21.3 ArduinoJson by Benoit Blanchon