Xinyuan-LilyGO/LilyGo-EPD-4-7-OWM-Weather-Display #9: Project not working on new board ESP32-S3 EPD47

volprecht, 2024-05-16:
Project crash after boot, display stop response. Will be any update for new board?

ant-munich, 2024-06-14:
The same here.
Board v2.4 2024-01-15.
With ESP32 V 2.0.15 and ArduinoJson up to 6.19.4 the Arduino IDE can compile, but according to serial monitor the board is constantly rebooting.

asg02fever, 2024-06-27:
for me after upload , nothing , screen is still at last image and serial show this :

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4209f31a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:

ugorur, 2024-07-18:
```
A14     : 0x00000059  A15     : 0x3ffb8a88  SAR     : 0x00000015  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4008b564  LEND    : 0x4008b56f  LCOUNT  : 0x00003f47  

Backtrace: 0x4008b561:0x3ffb21c0 0x400d352f:0x3ffb21d0 0x400d95e9:0x3ffb2230 0x400e2cd6:0x3ffb2290

ELF file SHA256: 5038d593d2e8bc69
```

Cpnclueless, 2024-11-29:
After uploading mine says
Leaving...
Hard resetting via RTS pin...

Cpnclueless, 2024-12-10:
Hey, has anyone solved the Guru meditation error yet?